insecurityofthings/uC_mousejack #1: Wiring scheme do not reflect the text.

LucaBongiorni, 2016-09-12:
Hey!
Cool implementation! I was thinking to make something like that with a Cactus Micro [1]
One thing I have noticed, from the prototype scheme are missing the LEDs wiring and also the capacitors.
Morevoer, about flashing the NRF24 I guess you flashed with:
platformio run --project-dir teensy-flasher --target upload

[1] https://www.tindie.com/products/AprilBrother/cactus-micro-rev2-arduino-compatible-plus-esp8266/

phikshun, 2016-09-13:
For the external LEDs, I de-soldered the on-board surface mount LEDs and soldered on small wires. Then I just hooked them up to a few external LEDs instead. I built another version with a clear box instead, to avoid that mess.

As for the caps, it's debatable if you even need them. I read somewhere that the NRF24L01 modules are pretty finicky with power stability. On the unit in the pic, I didn't use them... still works fine. But it doesn't hurt to put a 10uF and a 0.1uF in parallel between 3v and ground on the NRF module, just in case.

To upload the project, I used the IDE... Upload is an option in the "Platform IO" menu. Download the full Platform IO IDE, it's actually very pleasant to use. If you want to use the CLI, you'll just need to change the `--project-dir` option to use the uC_mousejack directory and you'll be good to go (AFAIK). I'm actually a Platform IO noob... I just found out about it while building a teensy programmer for a CrazyRadio PA I bricked.

Anyway, sorry about the bad documentation... I'll update with a full tutorial later this week.

Cactus Micro looks like a great board. It doesn't appear that Platform IO has explicit support for it, but according to the Cactus Micro page it says you can use "LilyPad USB" as the board type in the Arduino IDE. That's probably the case for Platform IO as well.

I'm working on two new versions, and I'll have separate branches for them. One will use the Feather Adalogger variant to load Duckyscripts from a microSD card, with the OLED display Featherwing to select the attack and the target. It will also have a USB type A connector and support classic USB HID injection. For the second, I'm going to try the Feather BLE variant, and build an iPhone app to build new attacks, remotely trigger, and for NRF24L01 targeting.

I considered Wifi, but it's inconvenient to switch over a phone to a different SSID just to interact with the device. You could configure the ESP8266 as a Wifi client instead, then have it call back for C2 comms, but that makes some assumptions about Wifi availability. BLE control seems simpler for my use cases.

If you have other ideas, let me know... always looking for cool projects. Feel free to fork, modify, etc.

LucaBongiorni, 2016-09-13:
Thanks for the info. Really helpful recommendations!
About... "One will use the Feather Adalogger variant to load Duckyscripts from a microSD card, with the OLED display Featherwing to select the attack and the target"
This is too cool! :) It reminds me when I was trying to create something like this [1] for wardriving a decade ago.

About Wifi, indeed you are right. It is not very convenient. I was considering it, due to the fact I am going to use the same hardware for Sensepost's USaBuse tool. 

[1] https://images-na.ssl-images-amazon.com/images/I/41Q6C0KHZSL._SX300_.jpg

phikshun, 2016-11-17:
Closing this issue out. I finally fixed the Fritzing :)

closed 2016-11-17 by phikshun