insecurityofthings/uC_mousejack #2: Adafruit substitute

LucaBongiorni, 2017-01-21:
Hi,
Having said I am not an arduino guru... Wondering how hard would be to port your aeduino code to a Arduino Mini Pro since it uses the same atmega pic of the adafruit's one?
I was thinking something like this one:
http://s.aliexpress.com/AFFBZfq2

phikshun, 2017-01-23:
Yes, but you'll need a 3.3v board instead of a 5v board (the NRF24 modules are 3.3v). Also, you'll need to change the defines for the CE and CSN pins to whatever pins your board of choice uses. Finally, you'll need to change the board in the platformio.ini file to match the board you want to use.

Definitely possible though, and shouldn't be too hard. All the remaining code should work just fine on anything ATmega32u4 based, and could be easily updated for other microcontrollers too.

LucaBongiorni, 2017-01-23:
Yeah, didn't notice was 5V. Thanks!
This should be perfectly fitting the need https://www.aliexpress.com/item/Free-shipping-Pro-Micro-Module-ATMEGA32U4-3-3V-8MHZ-with-2-row-pin-header-For-Leonardo/32233587710.html

closed 2017-01-23 by LucaBongiorni