insecurityofthings/uC_mousejack #3: Loop on "starting channel sweep"

LucaBongiorni, 2017-02-02:
Hi,
Said that I am testing against a Logitech K400r which is vulnerable and already proved with the CrazyRadioPA...
I built exactly your project (step by step).
It compiled and uploaded successfully with PlatformIO on the Feather.
But somehow is stuck in loop on "starting channel sweep".
Seems like is unable to find the keyboard around(despite the fact I am even generating traffic by typing with it).
I have even tried to different NRF24L01+ I bought.

Do you know a way I can see if the connection between Feather and NRF module is fine?
Is there a way I can communicate with th NRF module throught the serial and get some sort of ACK that it is alive?
Moreover, looking at your circuit scheme, I see the NRF's GND is not attached to the GND of the Feather. Might that be also an issue?

Any suggestion on how to troubleshoot my issue is more than welcome :)

Thanks mate!
Luca

minkione, 2017-02-03:
It was the GND. It must be connected between the boards ;]

LucaBongiorni, 2017-02-04:
The suggestion is right. GND has to be connected.

closed 2017-02-04 by LucaBongiorni