insecurityofthings/uC_mousejack #5: can't find any devices

M3m0r3X, 2017-05-30:
Hi,

i have tried to bild this project on a breadboard. The Firmware was compiled well and the serial monitor from platformio says in a loop: starting channel sweep
but it did not find any vulnerable device. I tried it wth different devices wich are vulnerable by "JackIt". Perhaps I did something wrong and perhaps you guys have an idea? I have even already setup an attackscript. Do I have to flash the nrf24 Module with rfstorm, too?

phikshun, 2017-06-08:
Can you give me a bit more detail about your setup? What board are you using and what module? What's the target device?

M3m0r3X, 2017-06-08:
I use all parts that are on this project:
- Adafruit Feather 32u4 protoboard
-  NRF24L01+ module (exact the same from the provided link)

The target devices are a Logitech M171, a Logitech k400 and a logiteck M325 as well. All of those where vulnerable by JackIt. I primary use the M171.

phikshun, 2017-06-11:
Resolved over Twitter. Closing issue.

closed 2017-06-11 by phikshun